mortbopet/Ripes #224: Malformed AppImage for v2.2.5

Irvise, 2022-09-02:
Hi,

thank you for the work that has gone into Ripes, it is quite fun to play around and learn with it.

I tried to download the latest Linux Appimage file ([Ripes-v2.2.5-linux-x86_64.AppImage.zip](https://github.com/mortbopet/Ripes/releases/download/v2.2.5/Ripes-v2.2.5-linux-x86_64.AppImage.zip) and it seems it is just a .zip file with the Windows file contents, not an actual Linux executable. I believe this is unintended behaviour.

Thank you for your time,
Fer

xdelu02, 2022-09-02:
Should be fixed now

mortbopet, 2022-09-02:
Yep, looks like something messed up when i uploaded the binaries for 2.2.5 - should be fixed!

Irvise, 2022-09-02:
I downloaded the new file and can confirm that it works now. I'll be closing the issue now.

Thank you!

closed 2022-09-02 by Irvise