mortbopet/Ripes #225: Not showing Pipeline table

shakthivels300, 2022-09-06:
[image]
Thank you so much for this software

mortbopet, 2022-09-06:
Does this still occur if your download the newest version (2.2.5)?

OBLIVIKRAM, 2022-09-08:
Yes it does

shakthivels300, 2022-09-08:
Yes sir we have downloaded it just last week

mortbopet, 2022-09-08:
Just to confirm, if you (in Ripes) go to `Help->Version` it says 2.2.5? Which OS are you running?

I am asking, since i am unable to reproduce the error on Windows, using 2.2.5 - for me, the pipeline diagram is generated correctly.

shakthivels300, 2022-09-09:
We uninstalled and installed fresh download it again it worked properly we were using lubuntu . May be some thing was gone wrong during previous download.
Thank you for your support

closed 2022-09-09 by mortbopet

jnoche, 2022-10-28:
FYI: This happened to one of my students who was using v2.2.4. When he replaced it with v2.2.5, the problem was fixed.