mortbopet/Ripes #233: Branching before Forwarding Unit in 5-Stage without Hazard Detection

MicahDoo, 2022-10-10:
Hi,
I noticed in the **5-Stage without Hazard Detection Processor**, the outputs from Register Read are pushed directly into the Branch Comparator without going through the Forwarding multiplexers.

That means whatever instructions involve the Branch Comparator will not get the benefit of forwarding. Am I right?

I tested it using the following code:

With `beq`, two stalls are needed after an `addi`:
```asm
.data
branch_str:
    .string "Branched.\n"
not_branch_str:
    .string "Not branched.\n"
    

.text

main:
    
    # a0 set to 10
    addi a0, zero, 10
    nop
    
    # a0 set to 0
    addi a0, zero, 0
    
    # stalls (less than 2 nops => error (no branch))
    # nop
    # nop
    
    # if a0 = 0, branch => if no error, should branch
    beq a0, x0, branch
    
not_branch:
    la a0, not_branch_str
    li a7, 4
    nop
    nop
    nop
    ecall
    
    j exit
        
branch:
    la a0, branch_str
    li a7, 4
    nop
    nop
    nop
    ecall
    
exit:
    
    # Exit
    li a7, 10
    nop
    nop
    nop
    ecall
```
With `addi`, no stalls are needed:
```asm
.data
branch_str:
    .string "Branched.\n"
not_branch_str:
    .string "Not branched.\n"
    

.text

main:
    
    # a0 set to 10
    addi a0, zero, 10
    nop
    nop
    
    # a0 set to 0
    addi a0, zero, 0
    
    # zero stalls
    
    # no hazard: a0 = 0 + 3 = 3 / hazard: a0 = 10 + 3 = 13
    addi a0, a0, 3
    
    li a7, 1
    nop
    nop
    nop
    ecall
    

exit:
    
    # Exit
    li a7, 10
    nop
    nop
    nop
    ecall
```

mortbopet, 2022-10-11:
Hi @MicahDoo 

So is the issue here not that there is a RAW hazard on the following?
```asm
    addi a0, zero, 0
    # if a0 = 0, branch => if no error, should branch
    beq a0, x0, branch
```
It is exactly such a situation which a hazard unit is supposed to detect and then dynamically stall the pipeline to realize the 2-cycle bubble (as you indicate with the two `nop` instructions).

The pipelined processors **without** hazard units are not supposed to execute a program correctly. They are meant as intermediate design points to be able to illustrate why hazard detection and resolution is required for a pipeline.  
It should probably be more clear in the program/processor selection dialog that these processors aren't guaranteed to produce correct results..!

MicahDoo, 2022-10-11:
Hi, @mortbopet
Thank you for the response!

Don't worry, the description was clear enough. I was just wondering why, with the "hazard detection off, forwarding on" option, forwarding seems to be incomplete in that it is removed for branching instructions, while kept intact for  ALU instructions. In my examples above, I didn't need any `nop` for RAW with the second instruction being `addi` because forwarding worked just fine for ALU but I did need two of them with `beq`.

And it shows on the datapath layout, too:
This is with hazard detection:
[image]
The ALU and Branch Comparator both take their inputs from the forwarding unit, as expected.

This is without hazard detection:
[image]
Here the Branch Comparator cannot receive write-back values coming through the two forwarding multiplexers because it doesn't receive inputs through them.

This threw me off a bit because, as far as I know, hazard detection happens in the control unit and is separate from forwarding, which is reflected on the datapath. I am curious as to why this configuration without hazard detection changes the forwarding functionality in the datapath in this way?

mortbopet, 2022-10-12:
Looking this through, i think you're right that there is a bug here - the branch unit should indeed be connected to the forwarding registers (as it is in the 5-stage model with fw/hz).

Once time permits, i'll push a fix.

MicahDoo, 2022-10-13:
Got it. I'm glad that's cleared up. This is an awesome tool and has helped me a lot with understanding RISC-V. Thank you for making this possible. Keep up the good work!