mortbopet/Ripes #237: Incorrect/inconsistent behaviour of mulhsu

StefMassin, 2022-10-20:
When writing tests for the multiply instructions of the RV32M extension, I noticed an inconsistency in the behaviour of the `mulhsu` instruction. It was found when running the code below. In the end, some value is expected in the registers t4 and s4. In practice however, they both end up being zero and nothing gets pushed to memory either. I also noticed in this case that the last 4 instructions appear as "Unknown Instruction" in the Processor view.

However, after removing any of the blocks under "Test2", s4 and t4 do end up with the correct values, which also get pushed to memory.

``` assembly
# Immediates
    li a0, 10
    li a1, -10
    li a2, 6
    li t0, 1
    li s0, 1

# Test1: mul, no mulh
while: # calculate 10^6 and (-10)^6
    beqz a2, break
    mul  t0, t0, a0
    mul  s0, s0, a1
    addi a2, a2, -1
    j    while
break:
    li a0, -1
    mul  s0, s0, a0 
    sw   t0, 104(zero) # 10^6    =  1.000.000 = 0x000F 4240
    sw   s0, 108(zero) # (-10)^6 = -1.000.000 = 0xFFF0 BDC0
    
# Test2: mul and mulh
    # Positive x positive
    mul  t1, t0, t0    # 10^6 * 10^6, lower bits
    mulh s1, t0, t0    # 10^6 * 10^6, upper bits
    sw   t1, 112(zero) # 0xD4A5 1000
    sw   s1, 116(zero) # 0x0000 00E8

    # Positive x negative
    mul  t2, t0, s0    # 10^6 * (-10)^6, lower bits
    mulh s2, t0, s0    # 10^6 * (-10)^6, upper bits
    sw   t2, 120(zero) # 0x2B5A F000
    sw   s2, 124(zero) # 0xFFFF FF17

    # Negative x negative
    mul  t3, s0, s0    # (-10)^6 * (-10)^6, lower bits
    mulh s3, s0, s0    # (-10)^6 * (-10)^6, upper bits
    sw   t3, 128(zero) # 0xD4A5 1000
    sw   s3, 132(zero) # 0x0000 00E8

# Test3: mul and mulhsu

    # Positive x negative, interpreted as positive
    # -1.000.000 = 0xFFF0 BDC0, so as a positive number, we have 4.293.967.296
    mul  t4, t0, s0    # 10^6 * "(-10)^6", lower bits
    mulhsu s4, t0, s0  # 10^6 * "(-10)^6", upper bits
    sw   t4, 136(zero) # 0x2B5A F000
    sw   s4, 140(zero) # 0x000F 4157

```

jnoche, 2022-10-23:
According to section 7.1 of "The RISC-V Instruction Set Manual Volume I: Unprivileged ISA" Document version 20191213, "If both the high and low bits of the same product are required, then the recommended code sequence is: MULH[[S]U] rdh, rs1, rs2; MUL rdl, rs1, rs2 (source register specifiers must be in same order and rdh cannot be the same as rs1 or rs2)." Have you tried swapping the MUL and MULH (or MULHSU) instructions (putting MULH or MULHSU before MUL)?

jnoche, 2022-10-23:
Also, in one of your comments you state "# (-10)^6 = -1.000.000 = 0xFFF0 BDC0". Shouldn't (-10)^6 be positive?

StefMassin, 2022-10-23:
> According to section 7.1 of "The RISC-V Instruction Set Manual Volume I: Unprivileged ISA" Document version 20191213, "If both the high and low bits of the same product are required, then the [...]

Swapping the instructions seems to have no effect. I get the same behaviour as before.

> Also, in one of your comments you state "# (-10)^6 = -1.000.000 = 0xFFF0 BDC0". Shouldn't (-10)^6 be positive?

Yes, almost every appearance of `(-10)^6` should actually be `-(10^6)`, that's my bad. However the program itself does generate 10^6 and -10^6 correctly, it is only the comments.

pjlobo, 2022-10-24:
Your code is modifiying itself.  The tests store their results starting at address 0x68 (104), which is where the first instruction of test 3 is located, so the instructions of test 3 are completely overwritten by tests 1 and 2. That's why you are seeing "Unknown instruction". Removing some instructions makes the program shorter and therefore the test 3 instructions aren't overwritten any more.

Assembler labeling is your friend. Here is a quick and dirty example using the [store global pseudo-instruction](https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md#load-and-store-global). More elegant solutions exist that avoid lots of unnecesary "auipc"s leveraging x3/gp.

```
# Immediates
    li a0, 10
    li a1, -10
    li a2, 6
    li t0, 1
    li s0, 1

# Test1: mul, no mulh
while: # calculate 10^6 and (-10)^6
    beqz a2, break
    mul  t0, t0, a0
    mul  s0, s0, a1
    addi a2, a2, -1
    j    while
break:
    li a0, -1
    mul  s0, s0, a0 
    sw   t0, test1r1, t6 # 10^6    =  1.000.000 = 0x000F 4240
    sw   s0, test1r2, t6 # (-10)^6 = -1.000.000 = 0xFFF0 BDC0
    
# Test2: mul and mulh
    # Positive x positive
    mul  t1, t0, t0    # 10^6 * 10^6, lower bits
    mulh s1, t0, t0    # 10^6 * 10^6, upper bits
    sw   t1, test2r1l, t6 # 0xD4A5 1000
    sw   s1, test2r1h, t6 # 0x0000 00E8

    # Positive x negative
    mul  t2, t0, s0    # 10^6 * (-10)^6, lower bits
    mulh s2, t0, s0    # 10^6 * (-10)^6, upper bits
    sw   t2, test2r2l, t6 # 0x2B5A F000
    sw   s2, test2r2h, t6 # 0xFFFF FF17

    # Negative x negative
    mul  t3, s0, s0    # (-10)^6 * (-10)^6, lower bits
    mulh s3, s0, s0    # (-10)^6 * (-10)^6, upper bits
    sw   t3, test3r1l, t6 # 0xD4A5 1000
    sw   s3, test3r1h, t6 # 0x0000 00E8

# Test3: mul and mulhsu

    # Positive x negative, interpreted as positive
    # -1.000.000 = 0xFFF0 BDC0, so as a positive number, we have 4.293.967.296
    mul  t4, t0, s0    # 10^6 * "(-10)^6", lower bits
    mulhsu s4, t0, s0  # 10^6 * "(-10)^6", upper bits
    sw   t4, 136(zero) # 0x2B5A F000
    sw   s4, 140(zero) # 0x000F 4157

end:
    li a7, 10
    ecall

.data
test1r1: .word 0
test1r2: .word 0
test2r1l: .word 0
test2r1h: .word 0
test2r2l: .word 0
test2r2h: .word 0
test3r1l: .word 0
test3r1h: .word 0
```

pjlobo, 2022-10-24:
> According to section 7.1 of "The RISC-V Instruction Set Manual Volume I: Unprivileged ISA" Document version 20191213, "If both the high and low bits of the same product are required, then the [...]

That's just a recommendation that makes it easier for a microarchitecture to identify whether instruction execution can be optimized (e.g. computing simultaneously both the high and low parts of the result), but the result must be the same regardless of order.

StefMassin, 2022-10-24:
> Your code is modifiying itself. The tests store their results starting at address 0x68 (104), which is where the first instruction of test 3 is located, so the instructions of test 3 are completely [...]

Yep, you're right, such a silly mistake. Thanks so much!

closed 2022-10-24 by StefMassin