mortbopet/Ripes #248: Prebuilt binary for Apple M1 silicon

rezoerme, 2022-11-19:
Do you have plans to release a prebuilt binary for Macs that have M1 silicon?

mortbopet, 2022-11-20:
I'd be open to a pull request for adding this to the CI build flow! I would expect this to come from an external contributor who has access to an M1 machine and can debug the flow and verify that the binary works as expected (which i do not).

mheese, 2022-12-15:
not sure if that helps at all, but at least I can confirm that building it and running it on M1 silicon does in fact work (commit `190418ad9555f5085be78acf8342159ec9a3f5d0`)

mortbopet, 2022-12-18:
Awesome! Again, i'd appreciate some help here from people who would actually use this release. Looking briefly into this, i'd expect that you could modify https://github.com/mortbopet/Ripes/blob/master/.github/workflows/mac-release.yml to perform a cross compilation (Github doesn't provide an aarch64 osx image, so we have to cross compile) with adjustments being:
- Pull down the correct prebuilt Qt version (for M1)
- Ensure that a clang version is available which has support for the `arm64-apple-macos11` target.
- Tell CMake to target aarch - might be enough to just add `-DC_FLAGS="-target arm64-apple-macos11" to the CMake command line

mheese, 2022-12-21:
We should probably actually try to build a universal binary (https://developer.apple.com/documentation/apple-silicon/building-a-universal-macos-binary).

@mortbopet I might be able to spare some cycles to make a PR which adjusts the github CI workflow, and I can also test the resulting binaries of course. However, I hope the expectations would not be that I'd be the maintainer for this?

mortbopet, 2022-12-29:
> However, I hope the expectations would not be that I'd be the maintainer for this?

No expectations. If things start breaking and there's no one available to fix it, then I'll disable the build. My hope would be that people who are interested in having an M1 binary available also have some vested interest in actually making that happen :).

atw1020, 2023-04-21:
Took a crack at this, and I think I succeeded. I can't open the app because it isn't signed, but I could run the executable. I want to try and make a universal installer, as @mheese suggested, but the homebrew version of qt6 I have is ARM only, and so I can't link it with the x86 version of the app. Trying to fix that now