mortbopet/Ripes #254: Ripes.exe cannot be restarted on Windows 10 after Ripes crashed

mistralino, 2022-12-13:
Ripes.exe cannot be restarted on Windows 10 after Ripes crashed. Even a reinstallation of Ripes and a reboot did not help. Any idea? Thanks.

mortbopet, 2022-12-14:
It might have to do with some invalid settings stored in the registry which thwarts Ripes starting up.

Try opening regedit and clear the settings under `HKEY_CURRENT_USER\Software\Ripes\Ripes` and see whether that changes anything.

mistralino, 2022-12-15:
...good point: this could explain why just deleting the subdir / rebooting is not sufficient.

mortbopet, 2023-01-02:
Will close this - fairly sure that it is related to the fix of https://github.com/mortbopet/Ripes/commit/5176a80b8fb3e1caf8e80b144a8f0cdb755336f9 and is resolvable by using the current continuous version + deleting the Ripes settings file.

closed 2023-01-02 by mortbopet