mortbopet/Ripes #255: Crashes when changing layout

atassi122, 2022-12-14:
Hi, 
The Program crashes after changing the layout from "standard" to "extended" on windows 11 22H2. 
And of course after that it woud not open until you go to registry editor and clear the settings,even rebooting won't help.

hypermenosh, 2022-12-23:
Same problem on windows 10 for Ripes v2.2.5

slinkish, 2023-01-02:
I believe this is also happening on the linux v2.2.5 AppImage binary, terminal output after switching to "extended" and similarly faulty setting is properly saved to `~./config/Ripes/Ripes.conf` so that the binary crashes immediately with the setting. 
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
zsh: IOT instruction (core dumped)  ./Ripes-v2.2.5-linux-x86_64.AppImage
```

mortbopet, 2023-01-02:
Fairly sure this is due to what was fixed by https://github.com/mortbopet/Ripes/commit/5176a80b8fb3e1caf8e80b144a8f0cdb755336f9. I'll release a new version of Ripes soon seeing as multiple people are encountering this.

mortbopet, 2023-01-07:
Went ahead and drafted up a new release https://github.com/mortbopet/Ripes/releases/tag/v2.2.6. Will close this issue for now since I'm fairly confident that the issue was resolved there. However, if not, feel free to re-open the issue.

closed 2023-01-07 by mortbopet