mortbopet/Ripes #260: Debugging breakpoint is not working

giraldeau, 2023-01-12:
It seems that adding a breakpoint to the assembly and running the code does not make the code pause at the instruction. Actually, the pause button does not work either.

Tested with the current master 976190de359dd683

[Screencast from 2023-01-12 05:46:55 PM.webm](https://user-images.githubusercontent.com/244101/212197966-985b5158-256b-411d-a4b0-a7be97888ac2.webm)

mortbopet, 2023-01-13:
I can reproduce the error - Hopefully i have time to dive into this this weekend.

giraldeau, 2023-01-13:
If you point me where to look at I can have a look.

Cheers!

giraldeau, 2023-03-01:
I tried to dig into the issue before checking your solution. Yeah, you were right, the auto clock was not stopped on breakpoint hit. As a quick fix I used the ProcessorHandler::stopping() signal. I confirm I tested your fix on the master branch and it is working just fine. Cheers!

```
diff --git a/src/processortab.cpp b/src/processortab.cpp
index a807966..f3d1474 100644
--- a/src/processortab.cpp
+++ b/src/processortab.cpp
@@ -232,6 +232,12 @@ void ProcessorTab::setupSimulatorActions(QToolBar *controlToolbar) {
   connect(m_autoClockTimer, &QTimer::timeout, this,
           [=] { ProcessorHandler::clock(); });
 
+  connect(ProcessorHandler::get(), &ProcessorHandler::stopping, [=]() {
+    if (m_autoClockTimer->isActive()) {
+      autoClock(false);
+    }
+  });
+
   const QIcon startAutoClockIcon = QIcon(":/icons/step-clock.svg");
   m_autoClockAction = new QAction(startAutoClockIcon, "Auto clock (F6)", this);
   m_autoClockAction->setShortcut(QKeySequence("F6"));
```

closed 2023-03-01 by giraldeau