mortbopet/Ripes #261: Application cannot be opened on macOS Ventura 13.1 (22C65)

JackWilli, 2023-01-13:
When trying to open version 2.2.6 on an x86 macOS machine I get the following message
`
The application cannot be opened for an unexpected reason, error=Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x600003e7c5a0 {Error Domain=NSPOSIXErrorDomain Code=111 "Unknown error: 111" UserInfo={NSLocalizedDescription=Launchd job spawn failed}}}
`

mortbopet, 2023-01-13:
hmm... and this doesn't happen for 2.2.5? As with everything macOS related, i have a hard time testing it myself seeing as i don't own a mac machine - and I thus just release the image that CI generates.
If you have the chance, it would also be helpful if you could build the top-of-main locally and see if that works... maybe something went wrong in macdeployqt.

gwentgod, 2023-03-27:
Encountered the same error on an apple silicon machine, and
```
chmod +x Ripes-v2.2.6-mac-x86_64.app/Contents/MacOS/Ripes
```
worked for me.

The local build works without any tweaking.

anta40, 2023-07-08:
> Encountered the same error on an apple silicon machine, and
> 
> ```
> chmod +x Ripes-v2.2.6-mac-x86_64.app/Contents/MacOS/Ripes
> ```
> 
> worked for me.
> 
> The local build works without any tweaking.

I'm still on Intel Mac (running Monterey). This works fine with the last stable release:
https://github.com/mortbopet/Ripes/releases/tag/v2.2.6

mortbopet, 2023-07-10:
Thank you for the heads up here - I've just added https://github.com/mortbopet/Ripes/commit/c36b9c76feffd2a015654956842b0e0fd26486b0 - would appreciate if a MacOS user could try out the continuous build once CI finished and verify that things work out of the box 🙏.

honi, 2023-07-11:
It now works out of the box on MacOS.

closed 2023-07-12 by mortbopet