mortbopet/Ripes #273: Is it possible to display the "Memory View" windows in the Editor tab?

luoxin13, 2023-04-13:
@mortbopet 
 
Thanks for your wonderful work on such an ISA simulator!

Since it is designed for teaching, I am wondering if it is possible to display the "Memory View" windows in the Editor tab.

In my opinion, such adjustment will benefit the user to observe more information while step-by-step debugging the assembly insts.

Since I am not familiar with UI-related development, I am wondering if the authors and other developers can provide some help?

I am expecting a Ripes software with the UI similar to:

[image]

mortbopet, 2023-04-13:
Hi @luoxin13 

It definitely should be possible, just a bit of tinkering required :). The main issue is that the code in Ripes is fairly heavily structured around the different tabs (which is not necessarily a good design point) so the memory viewer widget may be somewhat tangled into the other stuff that's going on in that tab. I don't think it's too much, though, and it should be possible to add what you're suggesting. I may get around to it this weekend :).

mortbopet, 2023-04-15:
Added - let me know what you think :).

luoxin13, 2023-04-17:
Thanks a ton! I have experimented with it. It is exactly what I wanted! 
BTW, maybe it can be further enhanced to support the F extension. :)

mortbopet, 2023-04-17:
F is a long-standing request, but most likely not something that I'm going to implement myself (a bit more involved since it would also need to affect the processor implementations). I'm hoping someone else would feel like this is a project worth pursuing since I've got a lot of other stuff on my plate nowadays :).

closed 2023-04-22 by mortbopet