mortbopet/Ripes #282: which version is compatible with Mac Mojave 10.14.6 ?

doremifajb, 2023-05-19:
Hello, I would like to install RIPES to my Macbook Pro mid 2012 with Mojave. Could you tell me which version is installable ? I tried a few newer versions but did not work. Thanks!

mortbopet, 2023-05-23:
Did you try one of the [continuous releases](https://github.com/mortbopet/Ripes/releases/tag/continuous)? Hopefully those should work with both old and new apple silicon, assuming that you have the latest OS. Apart from that, i don't do much to maintain releases for mac (seeing as i don't own a machine), and mainly defer this to other people who are interested in mac support. If those releases do not work, i think your best bet is to run the linux application under some form of virtualization.

doremifajb, 2023-05-23:
Thank you for the info! I just updated my Mac to Catalina and gave it a try, but none of the three files with "mac" worked... I have a win 8.1 under virtualbox, so I guess I'm going to try to use it there....

doremifajb, 2023-05-23:
Hello again, it didn't work with win 8.1, I guess it's also too old. But I have a Raspberry Pi 4B, so I downloaded  Ripes-v2.2.6-10-g4a49101-linux-x86_64.AppImage , did chmod u+x and double clicked it. I chose "execute". But nothing happened. Tried 
Ripes-v2.2.6-11-g91ceebb-linux-x86_64.AppImage but the same thing.... I thought Pi OS is a linux? Or is this not meant to work on RasPi?

doremifajb, 2023-05-23:
No... I don't have anything recent other than Raspberry pi;; and have no windows hardware. 
I just installed Win 10 on Virtualbox on my mac and tried with win version. After downloading a few dll files, I still get 
[image]
Much the same as with win 8.1... 

Is there any way to make it work on win 10? Or, better win 8.1 ?  I could also install Linux on Virtualbox, but since I need win anyway, it would be nice if I can get win version work. 
If not, what kind of linux should I get? 
I would appreciate your advice!

doremifajb, 2023-05-25:
Now I installed Ubuntu: ubuntu-22.04.2-desktop-amd64.iso on Virtualbox, but Ripes doesn't work there.... I tried both versions, made executable and run, but nothing happens.

mortbopet, 2023-05-26:
that sounds very strange - I'm sorry that you're having to go through these troubles to run the application, it usually works out of the box! On ubuntu, what happens if you run it through a terminal - does anything write to stdout?

doremifajb, 2023-05-26:
With ubuntu, there was another problem, that terminal didn't even run ! I installed another terminal (terminator or so), and then did chmod +x. But it still didn't work. Since several other applications didn't work either, I decided that perhaps ubuntu is too much for my macbook pro mid 2012 (dual core cpu, but upgraded 16GB RAM), I decided to try "lubuntu": light-version. lubuntu-22.04.2-desktop-amd64.iso (I picked one from https://linux.how2shout.com/6-best-linux-distros-to-try-on-virtualbox-virtual-machine/) 

 It still didn't work... 
Now I tried to run through a terminal (execute in terminal). Then a shell showed up with a message, "Are you sure you want to exit?". Regardless of if I click yes or no, nothing happens. 
On lubuntu, other things seem to work, so I think it is working fine (but upon booting it shows a message "failed to send log to host".) 
Is Ripes supposed to work with any regular linux ?

mortbopet, 2023-05-30:
Ideally, yes, seeing as it's distributed as an [appimage](https://appimage.org/).