mortbopet/Ripes #285: Going about adding a new processor

Anglo-software, 2023-06-04:
I'm currently working on extending the current 5-stage processor w/ hazard detection and forwarding to include a branch predictor, how would I go about adding this new processor to the registry? Also, is there an easier way to edit the layout of a processor other than editing the .json by hand?

Thanks!

closed 2023-06-05 by Anglo-software

mortbopet, 2023-06-06:
I assume you found https://github.com/mortbopet/Ripes/blob/master/docs/new_processor_models.md?

I've also hoped to see branch prediction in Ripes for a while now, so feel free to share your work in case you're planning on upstreaming this!

Anglo-software, 2023-06-06:
Actually I never really found that document lol. For adding the new processor I figured out that I forgot to add the new layouts to the `layouts.qrc` file (I was getting a seg fault before I asked the question), and for the editing of said layouts I scoured through the VSRTL code until I found the code that dealt with saving them, and then soon figured out that you can simply right click. Definitely a harder path to the same outcome but it did teach me a lot about how your software works.

Also I currently have the repo forked but once it passes the required tests I will most likely be upstreaming it!

Most likely it'll be some sort of correlated branch predictor (currently I have a (3,2) implementation going), but of course the code can be changed by the end user for whatever kind of prediction they'd like to simulate.

mortbopet, 2023-06-07:
Awesome, glad you found out how to add a layout, and also how things work under the hood.

Now, mind you, Ripes is/has been written over many years and has good, bad and ugly parts. So if you find something which makes no sense or is badly implemented, feel free to go to town in refactoring/improving things, as with all things FOSSy :).

Are you planning on adding a separate tab to configure and show information about the branch predictor?

Anglo-software, 2023-06-07:
Possibly, I'm not exactly the best with UI design (my focus is operating systems and low-level stuff), but I will still try to implement some mechanism to change the function of the branch predictor, whether it be in the UI or maybe a config file, still not sure. And if I don't implement the UI myself, there's always room for someone else to try.

Currently I'm still working on getting it to work in the first place. There's a lot of literature on how branch predictors make predictions, within their "black box" so to speak, but very little literature for the circuitry that is needed to drive it. Makes this as much invention as it is implementation.

Once I get it working then I'll try to figure out an interface for it. Have a good one!

Anglo-software, 2023-06-09:
Now that I have implemented my processor, how exactly do you run the unit tests? I've added the references to the new processor to the testing files noted in the document you shared, but I'm not exactly sure what to do from there.

Thanks!