mortbopet/Ripes #286: ECall to read input from stdin

tn4799, 2023-06-23:
Hello.
I've noticed that there are no ecalls to read input from stdin / the console. Did i miss something or is there no implementation for that? And if there is no ecall for reading input may you consider implementing ecalls for reading strings and numbers from stdin?

Have a nice day.

mortbopet, 2023-06-26:
This should definitely be there - try the `consoleReading.s` example which uses the '63' ecall (see `Help->System calls` inside Ripes).

closed 2023-09-25 by mortbopet

fmresearchnovak, 2024-03-12:
For me the demo hangs waiting for the user to enter the characters.  I tried pressing enter and many other keys but nothing seems to work.  What am I doing wrong?

[image]

fmresearchnovak, 2024-03-13:
I realized this is likely a duplicate of: [https://github.com/mortbopet/Ripes/issues/304](https://github.com/mortbopet/Ripes/issues/304)

neittien0110, 2024-03-16:
So do i. It always show the message " Waiting for user input..."
Opening the console of Web Browser, I found that this issue concerns **permissions** such as **clipboard**, **cookie**. 
![Clipboard permission] (https://github.com/mortbopet/Ripes/assets/8079397/2371b599-96a0-452d-86b9-4003f61d7b27)

Approving the clipboard permission. it still raises anothor warning. However, i cannot fix it. :( 
> QSettings::WebLocalStorageFormat requires cookies, falling back to IniFormat with temporary file"
put_char @ Ripes.js:9
write @ Ripes.js:9
write @ Ripes.js:9
doWritev @ Ripes.js:9
_fd_write @ Ripes.js:9
_emscripten_receive_on_main_thread_js @ Ripes.js:9
$func10551 @ Ripes.wasm:0xa45f3c
$func10540 @ Ripes.wasm:0xa45336
$jh @ Ripes.wasm:0x11e5ee8
ret.<computed> @ Ripes.js:9
executeNotifiedProxyingQueue @ Ripes.js:9
worker.onmessage @ Ripes.js:9