mortbopet/Ripes #287: Docker image build in Ubuntu 20.04 not working due to missing tag in dockerfile

kochaika, 2023-06-26:
Hi!
Is there any explicit purpose for using the base image when building a docker image without the tag? 
https://github.com/mortbopet/Ripes/blob/master/docker/ripes.dockerfile#L1

At the moment `latest` means `22.04`. But because of this, it is not possible to build an image on the Ubuntu 20.04 host OS. An error occurs:
```
Reading package lists...
E: Problem executing scripts APT::Update::Post-Invoke 'rm -f /var/cache/apt/archives/*.deb /var/cache/apt/archives/partial/*.deb /var/cache/apt/*.bin || true'
E: Sub-process returned an error code
The command '/bin/sh -c apt-get update -q     && apt-get install -qy --no-install-recommends     build-essential     cmake     gcc-riscv64-unknown-elf     git     libpthread-stubs0-dev     python3     python3-pip     libegl1 libegl1-mesa-dev     libgl1-mesa-dev libglib2.0-dev     libxkbcommon-x11-0 libpulse-dev libfontconfig1-dev libfreetype6-dev     libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-render-util0     libxcb-xinerama0 libxcb-composite0 libxcb-cursor0 libxcb-damage0     libxcb-dpms0 libxcb-dri2-0 libxcb-dri3-0 libxcb-ewmh2 libxcb-glx0     libxcb-present0 libxcb-randr0 libxcb-record0 libxcb-render0 libxcb-res0     libxcb-screensaver0 libxcb-shape0 libxcb-shm0 libxcb-sync1 libxcb-util1     && apt-get -y autoremove     && apt-get -y autoclean     && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
```

I suggest explicitly mentioning the base image tag in the docker file:
`FROM ubuntu:20.04`

In that case, the build will work on the Ubuntu 20.04 host OS as well

mortbopet, 2023-06-26:
This makes sense to me, feel free to submit a PR for the fix! 👍

kochaika, 2023-06-26:
Added fix to https://github.com/mortbopet/Ripes/pull/288

closed 2023-06-26 by mortbopet