mortbopet/Ripes #300: Improve UI colors by using system-provided palette

mortbopet, 2023-09-22:
Currently, Ripes doesn't really use the [system palette](https://doc.qt.io/qt-6/qapplication.html#palette) as made available through QApplication. A result of this is that colors throughout the application may look outdated/non-matching with respect to the system theme that the user is using on their system.

This can be improved by identifying places in Ripes where hardcoded colors are used, and replacing these in favor of using the system palette's provided colors (see the available colors in [QPalette](https://doc.qt.io/qt-6/qpalette.html)). 

Also relevant for: https://github.com/mortbopet/Ripes/issues/61
Note: this issue isn't restricted to Ripes, and mostly pertains to VSRTL, since that's where most of the drawing in Ripes is done.
More discussion on this topic: https://github.com/mortbopet/Ripes/issues/61

Monolitho, 2025-05-15:
Is this solved?

mortbopet, 2025-05-16:
> Is this solved?

Nope! It would be very nice to use some canonical way of doing application-wide themes, which i'm sure Qt provides (QPalette, Qt::ColorScheme, ...)