mortbopet/Ripes #310: Risc-V download

niccociampe, 2023-09-26:
Hello, I've tried a lot of times to download and run the programm,  but as I click the ripes.exe file nothing happens. It won't open, neither the program nor some sort of popup. My processor info are  : AMD Ryzen 7 3700U with Radeon Vega Mobile Gfx     2.30 GHz.    Is there someone who had can help me?

mortbopet, 2023-09-26:
what if you open the program through the console? Sometimes some information will be printed there.

niccociampe, 2023-09-26:
nothing happens, it doesn't print anything not even en error.

mortbopet, 2023-09-26:
I see - these cryptic issues tend to pop up from time and again. My best advice would be to try and download the source code and build ripes in debug mode, to see if that provides further clues. If you're not up for that, there's always the [web-based version](https://ripes.me/).

KidraMaster, 2023-11-21:
So, I had this problem and for me what worked was creating a new User in my PC without having special characters (in my case it was having an ã in the User name). Hope this helps.

closed 2023-12-04 by mortbopet