mortbopet/Ripes #326: Ripes crashes when using 'Open' Syscall without arguments

lasnikr, 2023-12-05:
Ripes crashes when using 'Open' Syscall without arguments:
```
li a7, 1024
ecall
```
I am on Mac and on the newest commit (built myself) but I also tried some older commits and saw the same bug (well, of course one should pass arguments but Ripes just crashing isn't what I would expect). I can't try with the latest release because it is broken on Mac.

raccog, 2023-12-09:
@lasnikr This seems to be fixed by #325, which should be pushed to the main branch soon.

#325 makes it so that (instead of crashing) ecall will return `-1` to the `a0` register, showing that it failed.

I'll keep this issue open until the PR is accepted into the main branch, so we can be sure that it fixes the bug.

raccog, 2023-12-21:
@mortbopet This issue should be closed; the provided example does not crash anymore on the main branch. It was fixed by #325.

closed 2023-12-22 by mortbopet