mortbopet/Ripes #327: Error reporting when running on centos7

gyx3598, 2023-12-06:
./Ripes-v2.2.6-35-gf6130db-linux-x86_64.AppImage: error while loading shared libraries: libOpenGL.so.0: cannot open shared object file: No such file or directory

Do you need to install qt6?
 It can be solved through the installation library

mortbopet, 2023-12-06:
That would be due to a missing system library ([a quick google search](https://stackoverflow.com/questions/65751536/importerror-libopengl-so-0-cannot-open-shared-object-file-no-such-file-or-dir) shows that this is probably solved if you apt-get `libopengl0` or `freeglut3`).

gyx3598, 2023-12-09:
I installed these two packages. It seems that the original running error has not been solved

gyx3598, 2023-12-09:
> That would be due to a missing system library ([a quick google search](https://stackoverflow.com/questions/65751536/importerror-libopengl-so-0-cannot-open-shared-object-file-no-such-file-or-dir) shows that this is probably solved if you apt-get or ).`libopengl0``freeglut3`

I installed these two packages. It seems that the original running error has not been solved

mortbopet, 2025-01-05:
Closed due to inactivity

closed 2025-01-05 by mortbopet