mortbopet/Ripes #350: "Clock the circuit with the selected frequency" (F6) never stops

gamberoillecito, 2024-03-18:
Ripes version: v2.2.5-10-g976190d
Bug description: running any program pressing the green arrow leaves the program in a state where it is always "re-run" from the beginning as soon as the .s file is edited. Furthermore most of the controls stop working.

After pressing the green arrow the only way I found to get the program back to normal is to restart it.

See the video below

https://github.com/mortbopet/Ripes/assets/42670032/92e1804c-d266-4eaa-9f14-90c400622657

stevenbell, 2024-03-26:
I'm observing the same problem!  I was able to hit pause while the code was in a loop to get it out of this behavior, but for straight-through code this isn't possible.

gamberoillecito, 2024-03-26:
@stevenbell the best solution I found so far is the following (sort of GTA cheat code, i.e. you need to press the buttons really fast):
"<" 🔄🔄⏸️⏸️🔄
I was able to reset the processor even for a program with 4 instructions @ 300ms

mortbopet, 2024-04-02:
I'm able to reproduce this for v2.2.6, but it doesn't seem to reproduce for the [continuous release](https://github.com/mortbopet/Ripes/releases/tag/continuous) nor on [WASM](https://ripes.me/) so it might have been fixed in the meantime. Would you mind trying out one of those to check if the issue is still there?

stevenbell, 2024-04-02:
Indeed, it seems to be fixed in g027e678 !

gamberoillecito, 2024-04-07:
@mortbopet it seems to be solved in 8780873

closed 2024-04-07 by gamberoillecito