mortbopet/Ripes #355: Assembly source code execution highlight only works with default .text start address

Aurelio-AVB, 2024-04-02:
Hi,

I was doing some testing and noticed that the source code execution highlight only appears when the start address for the .text section is the default (0x00000000).

[image]

If I change the .text start address source code highlight does not appear 
[image]

I am using ripes 2.2.6-41-gb71f0dd on Win11.

Thanks for any input and regards,
Aurelio Villalpando

mortbopet, 2024-04-02:
Thank you for reporting this; i'm sure this'll be an easy thing to fix - suspecting that the PC is used to index the text blocks in the code editor, but we're never doing the math to also account for a non-zero starting address of the code. Will look into it asap!

mortbopet, 2024-04-02:
@Aurelio-AVB should be fixed now; please let me know if it works for you, and if not, feel free to reopen.

commit 027e678a (2024-04-02): Account for non-zero starting address when recording source mappings

Fixes #355
closed 2024-04-02 by mortbopet