mortbopet/Ripes #358: .gitignore does not match files generated at configuration/compile time (?)

silvigon, 2024-04-22:
Hi, opening a different issue for this since it's not directly related to the branches project (which has slowly been progressing, despite my probably unwise choice for a first C++ project 🙃), but has made the development workflow a bit fiddly at times. Now, given the previous information, it's very possible that I'm just missing something, but it doesn't seem to have been brought up in an issue before and it hasn't been my experience working on software projects in other languages, so:

Shouldn't .gitignore catch the files that get generated when configuring and compiling the project? I mean, I know it already catches some of them, and it wasn't too difficult to add the ones it didn't for my copy of the repo, but it feels like a weird "default behaviour" for the project to have a bunch of files muddying the `git status` output like that.

mortbopet, 2024-06-03:
Typically, in-tree builds are expected to be in some `build*`-named folder. Hence, everything output in this folder will be captured by [this line]https://github.com/mortbopet/Ripes/blob/master/.gitignore#L6) in the gitignore. If you for some reason built your project in a folder inside the repo not prefixed by `build` _or_ you've run `cmake` with the base repo as the build directory, then you'll get the `git` muddying.

silvigon, 2024-06-12:
I probably should've thought of that, now that you mention it 🙃

Thanks!

closed 2024-06-12 by silvigon