mortbopet/Ripes #371: Building from source fails on Fedora 40

silvigon, 2024-10-03:
Was able to reproduce this on the live environment of a Fedora 40 image, where building according to the instructions produces the following error:
```
In file included from /home/liveuser/Ripes/external/VSRTL/external/cereal/include/cereal/archives/json.hpp:56,
                 from /home/liveuser/Ripes/external/VSRTL/graphics/vsrtl_componentgraphic.cpp:14:
/home/liveuser/Ripes/external/VSRTL/external/cereal/include/cereal/external/rapidjson/document.h: In member function ‘rapidjson::GenericStringRef<CharType>& rapidjson::GenericStringRef<CharType>::operator=(const rapidjson::GenericStringRef<CharType>&)’:
/home/liveuser/Ripes/external/VSRTL/external/cereal/include/cereal/external/rapidjson/document.h:319:82: error: assignment of read-only member ‘rapidjson::GenericStringRef<CharType>::length’
  319 |     GenericStringRef& operator=(const GenericStringRef& rhs) { s = rhs.s; length = rhs.length; }
      |                                                                           ~~~~~~~^~~~~~~~~~~~
make[2]: *** [external/VSRTL/graphics/CMakeFiles/vsrtl_gfx_lib.dir/build.make:137: external/VSRTL/graphics/CMakeFiles/vsrtl_gfx_lib.dir/vsrtl_componentgraphic.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:990: external/VSRTL/graphics/CMakeFiles/vsrtl_gfx_lib.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

silvigon, 2024-10-07:
Perhaps a similar issue to https://github.com/bambulab/BambuStudio/issues/4018 ?

silvigon, 2025-02-25:
As suggested by the mentioned issue, installing gcc13 and configuring CMake to use it (`cmake -DCMAKE_CXX_COMPILER=gcc-13`) does allow compilation to advance further, but then the linker fails. Hopefully I'm just missing something?
```
[ 60%] Linking CXX executable Ripes
/usr/bin/ld: external/VSRTL/graphics/libvsrtl_gfx_lib.a(vsrtl_componentgraphic.cpp.o): undefined reference to symbol '_ZNSt14basic_ifstreamIcSt11char_traitsIcEEC1ERKNSt7__cxx1112basic_stringIcS1_SaIcEEESt13_Ios_Openmode@@GLIBCXX_3.4.21'
/usr/bin/ld: /usr/lib64/libstdc++.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/Ripes.dir/build.make:323: Ripes] Error 1
make[1]: *** [CMakeFiles/Makefile2:626: CMakeFiles/Ripes.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

silvigon, 2025-02-27:
I've now tried just going into the cereal submodule and updating it to v1.3.2, and that seems to work just fine with gcc14 👍🏻

mortbopet, 2025-03-02:
So, sounds like we need a Cereal bump! I've got some stale work on refactoring cmake in both VSRTL and Ripes, but this seems like a good motivation to get some of that stuff finished off...!

mortbopet, 2025-03-02:
https://github.com/mortbopet/Ripes/pull/395 bumped the VSRTL version - which should have set the cereal version to the current most-recent commit. Try to pull main, and let me know how it works :).

silvigon, 2025-03-02:
Works like a charm! Closing.