mortbopet/Ripes #377: Counting cycles from one instead of zero?

silvigon, 2024-11-07:
An issue that was brought up in my latest meeting about the project described in #349 (which, I'm happy to report, is progressing nicely) is that students are routinely confused by Ripes counting processor cycles from zero instead of one, and my professors have asked me to investigate whether this could be changed.  

I've come up with an initial implementation for this that seems to work, but I'd like to know if there's a reason for the default behaviour that we're missing.

mortbopet, 2024-12-02:
It's a good question... hmm, I don't think any grander thoughts have been put behind why it starts from 0, but now that i think about it, i can see the confusion. The initial state of the processor is one where the instruction has already been loaded from instruction memory - regardless of processor model. Which somewhat implies that a clock cycle has occurred to make that load occur - do you agree? If we agree, I think that's a good argument for start at 1!

silvigon, 2024-12-11:
Sounds good to me! If it's something you're interested in for mainline Ripes, I'll do a pull request with my implementation as soon as I get the green light to host a mirror of our repo here.  

Some of the other features we're working on are also either done (additional processor models from #349, improved "About" dialog), or in progress (modified processor selection dialog); let me know if you'd be interested in getting any of these merged as well :>