mortbopet/Ripes #378: docker container build fails

PatrizioColomba, 2024-11-13:
building ripes using docker file fails. Adding python3-dev package solved the issue on my side.

Here the failure log

```
[3/5] RUN python3 -m pip install aqtinstall
	Collecting aqtinstall
	Downloading aqtinstall-3.1.18-py3-none-any.whl (61 kB)
	Collecting defusedxml
	Downloading defusedxml-0.7.1-py2.py3-none-any.whl (25 kB)
	Collecting bs4
	Downloading bs4-0.0.2-py2.py3-none-any.whl (1.2 kB)
	Collecting py7zr>=0.20.6
	Downloading py7zr-0.22.0-py3-none-any.whl (67 kB)
	Collecting texttable
	Downloading texttable-1.7.0-py2.py3-none-any.whl (10 kB)
	Collecting semantic-version
	Downloading semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)
	Collecting humanize
	Downloading humanize-4.10.0-py3-none-any.whl (126 kB)
	Collecting requests>=2.31.0
	Downloading requests-2.32.3-py3-none-any.whl (64 kB)
	Collecting patch>=1.16
	Downloading patch-1.16.zip (12 kB)
	Collecting beautifulsoup4
	Downloading beautifulsoup4-4.12.3-py3-none-any.whl (147 kB)
	Collecting psutil; sys_platform != "cygwin"
	Downloading psutil-6.1.0-cp36-abi3-manylinux_2_12_x86_64.manylinux2010_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (287 kB)
	Collecting inflate64<1.1.0,>=1.0.0
	Downloading inflate64-1.0.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (94 kB)
	Collecting pycryptodomex>=3.16.0
	Downloading pycryptodomex-3.21.0-cp36-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (2.3 MB)
	Collecting multivolumefile>=0.2.3
	Downloading multivolumefile-0.2.3-py3-none-any.whl (17 kB)
	Collecting pyppmd<1.2.0,>=1.1.0
	Downloading pyppmd-1.1.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (139 kB)
	Collecting brotli>=1.1.0; platform_python_implementation == "CPython"
	Downloading Brotli-1.1.0-cp38-cp38-manylinux_2_5_x86_64.manylinux1_x86_64.whl (2.8 MB)
	Collecting pybcj<1.1.0,>=1.0.0
	Downloading pybcj-1.0.2-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (50 kB)
	Collecting pyzstd>=0.15.9
	Downloading pyzstd-0.16.2.tar.gz (789 kB)
	Installing build dependencies: started
	Installing build dependencies: finished with status 'done'
	Getting requirements to build wheel: started
	Getting requirements to build wheel: finished with status 'done'
	Preparing wheel metadata: started
	Preparing wheel metadata: finished with status 'done'
	Collecting idna<4,>=2.5
	Downloading idna-3.10-py3-none-any.whl (70 kB)
	Collecting urllib3<3,>=1.21.1
	Downloading urllib3-2.2.3-py3-none-any.whl (126 kB)
	Collecting certifi>=2017.4.17
	Downloading certifi-2024.8.30-py3-none-any.whl (167 kB)
	Collecting charset-normalizer<4,>=2
	Downloading charset_normalizer-3.4.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (143 kB)
	Collecting soupsieve>1.2
	Downloading soupsieve-2.6-py3-none-any.whl (36 kB)
	Building wheels for collected packages: patch, pyzstd
	Building wheel for patch (setup.py): started
	Building wheel for patch (setup.py): finished with status 'done'
	Created wheel for patch: filename=patch-1.16-py3-none-any.whl size=12407 sha256=141fdbe7cf982136065b25ac56281b7c5af0e6a0952f5be8650e271d62dc2c39
	Stored in directory: /root/.cache/pip/wheels/ee/1f/d7/98669d171c98079d257b42489cd3781307dd4831bcbc8d840a
	Building wheel for pyzstd (PEP 517): started
	Building wheel for pyzstd (PEP 517): finished with status 'error'
	ERROR: Command errored out with exit status 1:
	command: /usr/bin/python3 /tmp/tmpenip_mc3 build_wheel /tmp/tmpan5me_ft
	cwd: /tmp/pip-install-oipiqlg9/pyzstd
	Complete output (49 lines):
	running bdist_wheel
	running build
	running build_py
	creating build
	creating build/lib.linux-x86_64-cpython-38
	creating build/lib.linux-x86_64-cpython-38/pyzstd
	copying src/zstdfile.py -> build/lib.linux-x86_64-cpython-38/pyzstd
	copying src/seekable_zstdfile.py -> build/lib.linux-x86_64-cpython-38/pyzstd
	copying src/__init__.py -> build/lib.linux-x86_64-cpython-38/pyzstd
	copying src/__main__.py -> build/lib.linux-x86_64-cpython-38/pyzstd
	creating build/lib.linux-x86_64-cpython-38/pyzstd/c
	copying src/c/__init__.py -> build/lib.linux-x86_64-cpython-38/pyzstd/c
	copying src/__init__.pyi -> build/lib.linux-x86_64-cpython-38/pyzstd
	copying src/py.typed -> build/lib.linux-x86_64-cpython-38/pyzstd
	running build_ext
	+--------------------------------------------+
	|             Pyzstd build config            |
	+-------------------------+------------------+
	| Pyzstd version          | 0.16.2           |
	+-------------------------+------------------+
	| Implementation          | C                |
	+-------------------------+------------------+
	| Link to zstd library    | Statically link  |
	+-------------------------+------------------+
	| Enable AVX2/BMI2        | False            |
	+-------------------------+------------------+
	| Debug build             | False            |
	+-------------------------+------------------+
	| Warning as error        | False            |
	+-------------------------+------------------+
	building 'pyzstd.c._zstd' extension
	creating build/temp.linux-x86_64-cpython-38
	creating build/temp.linux-x86_64-cpython-38/src
	creating build/temp.linux-x86_64-cpython-38/src/bin_ext
	creating build/temp.linux-x86_64-cpython-38/zstd
	creating build/temp.linux-x86_64-cpython-38/zstd/lib
	creating build/temp.linux-x86_64-cpython-38/zstd/lib/common
	creating build/temp.linux-x86_64-cpython-38/zstd/lib/compress
	creating build/temp.linux-x86_64-cpython-38/zstd/lib/decompress
	creating build/temp.linux-x86_64-cpython-38/zstd/lib/dictBuilder
	x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DPYZSTD_STATIC_LINK -DZSTD_MULTITHREAD -Izstd/lib -Izstd/lib/common -Izstd/lib/dictBuilder -I/usr/include/python3.8 -c src/bin_ext/pyzstd.c -o build/temp.linux-x86_64-cpython-38/src/bin_ext/pyzstd.o -g0 -flto
	In file included from src/bin_ext/pyzstd.c:1:
	src/bin_ext/pyzstd.h:8:10: fatal error: Python.h: No such file or directory
	8 | #include "Python.h"
	|          ^~~~~~~~~~
	compilation terminated.
	/tmp/pip-build-env-t251jhlw/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py:268: UserWarning: Unknown distribution option: 'test_suite'
	warnings.warn(msg)
	error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
	----------------------------------------
	ERROR: Failed building wheel for pyzstd
	Successfully built patch
	Failed to build pyzstd
	ERROR: Could not build wheels for pyzstd which use PEP 517 and cannot be installed directly
`

mortbopet, 2024-12-02:
Thanks for the suggested fix - added in https://github.com/mortbopet/Ripes/blob/master/docker/ripes.dockerfile.

PatrizioColomba, 2024-12-03:
Thanks for the update, glad to help. Looking forward to the next release.

closed 2024-12-10 by mortbopet