mortbopet/Ripes #392: Disassembler does not correct dissasemble register for compressed instructions

mortbopet, 2025-02-03:
> Disassembler does not correct dissasemble register, when reduce range register x8-x15
> and not error for  reduce reg in asm source.
> ( Asm and C  mode )
> [image]
> 
> Command  c.beqz  and  c.bnez   not  valid  compute  offset.
> ( Asm mode )
> 
> format command   c.lw and  c.sw
> c.lw  rd,  rs,  offset
> 
> not really   
> c.lw   rd,  offset(rs) 

 _Originally posted by @Zikon328 in [#125](https://github.com/mortbopet/Ripes/issues/125#issuecomment-2630162935)_

mortbopet, 2025-03-29:
@Zikon328 thank you for raising the issue; this should now have been fixed in https://github.com/mortbopet/Ripes/commit/40ebba770de4342eb9f5af6e0ca037284aaa4966.

closed 2025-03-29 by mortbopet

Xiamatsu, 2025-05-21:
Branch and jump commands "C"  instructions is not valid offset !
c.beqz c.bnez
c.j   c.jal
The absolute address is used, not relative.

Commands  c.lw and c.sw   -  used only format
c.lw  rd, rs, offset

can do  [ c.lw  rd,  (offset)rs ]  ?